emilio1625/Anycubic-Kossel-Marlin #2: Z height

magloviti, 2018-01-15:
[Marlin.zip](https://github.com/emilio1625/Anycubic-Kossel-Marlin/files/1631887/Marlin.zip)

hi,
My Z Height is stuck at 238.54mm after homing, should be around 290,
I changed Value on printer to 290mm, And when I reed data about printer after connecting it to Pronterface it says, among other data:
"echo:Delta settings: L<diagonal_rod> R<radius> H<height> S<segments_per_s> B<calibration radius> XYZ<tower angle corrections>
echo:  M665 L218.00 R101.31 H290.00 S160.00 B80.00 X0.00 Y0.00 Z0.00"
 Everything, except that, seams to work ok, but I 'm not sure  because I can't test with nozzle at +50mm from bed.

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

emilio1625, 2018-01-15:
you must set the correct delta height in the firmware, it must be around +/- 50mm the real height, all the calibration routines use the delta height value defined in the firmware, and doesn't obey the delta height in the eeprom or sram, that's not a bug, it's a (safety) feature! xD

magloviti, 2018-01-15:
I just did that  define_delta_height 290
Also changed temperature to start extruder fan from 50 to 18°C,
and fan starts at 18, so I know that I have good firmware, but height stays same at 238.54

magloviti, 2018-01-15:
OK, its fixed now. Ended up with Delta height of 350mm, and after upload I sent M502, and M500 from pronterface.

closed 2018-01-15 by magloviti