emilio1625/Anycubic-Kossel-Marlin #3: Open a PR to merge the changes back into Marlin

archi, 2018-02-20:
Thanks for your work, my printer runs really well with this fork. Even with the TMC2130.

I am just surprised why you single-handedly maintain a custom fork instead of having your changes merged back into the firmwares master branch? I'd be surprised if they/@thinkyhead wouldn't accept most of your changes (especially since it's difficult to switch to their current bugfix branch, or 2.x...).

emilio1625, 2018-02-23:
Hi, thanks :)
I actually don't see the point to open a PR, i have not added any new feature, i just made more simple the printer configuration, mostly for noobs, I'm sure any power user would find my changes redundant. 
I would open a PR for supporting the TRIGORILLA board, but there is one already.
But if you have any idea on what exactly could be PR'ed i could try. Maybe i could maintain a default config for the printer

thinkyhead, 2018-02-23:
If you feel like adding an Anycubic Kossel to the main repository, we'd welcome it. Currently our delta selection is rather meagre:
[image]

gotdirt24, 2018-03-06:
I’m trying to add the tmc2130 on the xyz to the trigorilla board I’m getting compile errors after installing library for tmc2130 , as far as editing the pins I have no idea on how to do it and which pins to move.

emilio1625, 2018-03-08:
I don't own the TMC2130, but i think they use i2c to communicate with the board, the i2c pins are connected to the lcd header i think, i could try to do some research about it, but i don't know if i can give a good support to all the features of these drivers

emilio1625, 2018-03-08:
@thinkyhead what do you need me todo to add the anycubic to the Marlin repo? Would you like a folder for every Anycubic Kossel version? there are currently 3 really different versions, what features can i enable by default? or can i just do a PR with the config file that i have already been working on? PD: I have already read your contributing guides.

archi, 2018-03-08:
@gotdirt24 you should perhaps open a new github issue for your issue instead of hijacking an existing, unrelated issue ;-) That's generally frowned upon on most bug trackers, but this is just a friendly reminder. Since the [watterott] TMC2130 work great for me I would be glad to help you out, but I need more information. For example your configuration files and some more details on how you wired them up to the board [I did disconnect the SD card/rotary encoder cable to get access to the SPI pins]).
Just open a new issue with that info contained and ping me so I don't miss it.

archi, 2018-03-08:
@emilio1625 regarding what could be PR'd: The anycubic board file, it's integration in the existing configuration infrastructure and the example configurations would be most useful, I think.
I didn't check all your commits (I'm in the midst of moving and have too much real life stuff to do), so I don't know if you did any source changes that might need merging.

thinkyhead, 2018-03-11:
> can i just do a PR with the config file that i have already been working on?

@emilio1625 — Starting with a fresh copy of the current `bugfix-1.1.x` branch…
```
git fetch upstream
git checkout upstream/bugfix-1.1.x -b bf1_anycubic_kossel
```

- Create a folder: `example_configurations/delta/Anycubic/Kossel`
- Copy the `delta/generic/Configuration.h` and `Configuration_adv.h` into this folder.
- Git add and git commit the unmodified files.
- Apply your changes to these files.
- Use Github Desktop to cherry-pick and commit your changes.
- Submit a PR targeting `bugfix-1.1.x`.

Then do the same for `bugfix-2.0.x`, except the delta configs are in `src/config/examples/delta`.