emilio1625/Anycubic-Kossel-Marlin #4: Printer stops all function after first few layers. Requires restart

jar0fair, 2018-02-22:
##Bug Report
- Description: Printer is freezing up and stops all fans and motion after first layer or two of printing. LCD screen is unresponsive, requires restart.
- Expected behaviour: Complete the print
- Actual behaviour: print aborted, machine unresponsive.
- Steps to reproduce: “Print from SD”

emilio1625, 2018-02-23:
have you tried with another SD card? does this happens also when printing by serial interface?

jar0fair, 2018-02-23:
@emilio1625 Yes, I have printed with multiple SD cards and have attempted to print files that were printed successfully in the past, from both SDs and the issue persists. I am wondering if the PSU has failed.

jar0fair, 2018-03-09:
@emilio1625 Wow, interesting. I just tried it, and it did print okay via USB. Oh great...now I have to troubleshoot the LCD screen