emilio1625/Anycubic-Kossel-Marlin #6: Unable to compile

marcottt, 2018-04-01:
### Description
unable to compile firmware in ide (1.8.1) with file from delta\anycubic folder

Arduino:1.8.1 (Windows 7), Scheda:"Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch\MarlinConfig.h:40:0,

                 from D:\dropbox\3d\firmware\marlin originale\github copia\Anycubic-Kossel-Marlin-1.1.x\Marlin\Marlin.ino:31:

SanityCheck.h:876: error: #error "LCD_BED_LEVELING requires MESH_BED_LEVELING or ABL with PROBE_MANUALLY."

     #error "LCD_BED_LEVELING requires MESH_BED_LEVELING or ABL with PROBE_MANUALLY."

      ^

exit status 1
#error "LCD_BED_LEVELING requires MESH_BED_LEVELING or ABL with PROBE_MANUALLY."

### Steps to Reproduce

1. copy all folder locally
2. copied configuration.h and configuration.adv.h in Marlin folder from example_configurations\delta\anycubic
3. open Marlin.in with IDE 
4. try compile.

**Expected behavior:** [What you expect to happen]
compile it

**Actual behavior:** [What actually happens]
error as above

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* Read the "Contributing" page (under "Helpful resources") for additional guidelines.

emilio1625, 2018-04-01:
should be fixed by a60fb6f6e9918b1a6e4748ec71de6aaa6241d8f6, please confirm

marcottt, 2018-04-02:
Yes compilation is fixed now. thanks

closed 2018-04-02 by marcottt