emilio1625/Anycubic-Kossel-Marlin #9: Cooling FAN for filament

jscurtu, 2018-06-13:
I Installed this firmware to my Anycubic Kossel Linear Plus (upgraded).
I am havining a issue with the part cooling fan, it is not turning on when I start my print. 
Do you know what I might be doing wrong? I looked in all settings in Configuration.h but I coulndt find anything.

When I print I always have to go manually in the printer and crank the speed value of "FAN speed 1" up to 255.

Thanks

closed 2018-06-14 by jscurtu

SergioSV96, 2020-03-29:
Did you find a solution? I am having the same problem, thank you.

SergioSV96, 2020-03-29:
> I Installed this firmware to my Anycubic Kossel Linear Plus (upgraded).
> I am havining a issue with the part cooling fan, it is not turning on when I start my print.
> Do you know what I might be doing [...]

Did you find a solution? I am having the same problem, thank you.

xarbit, 2020-03-29:
Yes because I used my printer alot, but honestly I cannot remember if it was because the trigorilla board was broken or because I created my own firmware. I had a few burnt mosfets and switched the board a few times. 

https://github.com/xarbit/Marlin_119_Anycubic_Kossel_Linear_Plus